ARMmbed/esp8266-driver #3: compiling fails to ATParser.cpp:328

jupe, 2016-07-04:
Got this:

```
mbed compile --build .build/K64F_iar_arm_default/ -m K64F -t IAR -c 
Building project mbed-os-cliapp (K64F, IAR)
Compile: at24mac.cpp
Compile: driverAtmelRFInterface.cpp
Compile: driverRFPhy.c
Compile: ESP8266Interface.cpp
Compile: ATParser.cpp
[Error] ATParser.cpp@328: [Pe029]: expected an expression
      _oobs.push_back((struct oob){strlen(prefix), prefix, cb});
                                  ^
"Z:\builds\ws\ARMmbed\mbed-os-cliapp\nsapi_changes\mbed-os-cliapp\esp8266-driver\ESP8266\ATParser\ATParser.cpp",328  Error[Pe029]: expected an expression
[ERROR] 
      _oobs.push_back((struct oob){strlen(prefix), prefix, cb});
                                  ^
"Z:\builds\ws\ARMmbed\mbed-os-cliapp\nsapi_changes\mbed-os-cliapp\esp8266-driver\ESP8266\ATParser\ATParser.cpp",328  Error[Pe029]: expected an expression

[mbed ERROR] "python" returned error code 1.
```

geky, 2016-07-04:
Ah, good catch. I believe this is another case of gnu-extensions letting things through.

I have a fix on https://github.com/ARMmbed/esp8266-driver/tree/fix-iar, but I won't be able to test it until I get back to the office.

geky, 2016-07-05:
Should be fixed by #4, feel free to reopen if this is not the case.

PR #4 (2016-07-05): Removed use of C struct initializer in C++
Fixes #3

Current flags for mbed-os let gnu-extensions into c++ code. This issue is documented here:
https://github.com/ARMmbed/mbed-os/issues/241

This removes C struct initializers, a gnu-extension.
closed 2016-07-05 by geky