ARMmbed/esp8266-driver #7: ESP8266-driver master does not compile against mbed OS 5.1.2

janjongboom, 2016-08-30:
https://github.com/ARMmbed/esp8266-driver/commit/450cc128865ffb90b5cbfe5af193621644024fa7 does...

But when someone adds esp8266-driver to their project they get the wrong one.

@geky

janjongboom, 2016-08-30:
Error with ARMCC (not with GCC_ARM):

```
Error: Object of abstract class type "ESP8266Interface" is not allowed in "easy-connect/easy-connect.h", Line: 17, Col: 19
```

geky, 2016-08-30:
Hi @janjongboom 

We have been keeping the esp8266 master compatible with the mbed-os master, and there should be no issues using the most recent versions of both.

You are right that the current mbed-os release has fallen out of sync with the esp8266, but unfortunately the master of the esp8266 can't be kept compatible with both mbed-os release and mbed-os master. The next release of mbed-os should resolve this specific issue.

To hopefully alleviate some of these problems, I've started creating releases for the esp8266. The current release is compatible with [`v1.0`](https://github.com/ARMmbed/esp8266-driver/releases/tag/v1.0).

closed 2016-09-08 by janjongboom