ARMmbed/esp8266-driver #8: Only works with very particular version of Espressif firmware

janjongboom, 2016-08-30:
Have been trying this driver with mbed Client against recent versions of Espressif (1.5 and 2.0)... Basic stuff works fine like joining the network, but then...

```
AT> AT+RST
AT< AT+RST
AT< 
AT= OK
AT< 
AT< WIFI DISCONNECT
AT< 
AT<  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
AT< 
AT< load 0x40100000, len 1856, room 16 
AT< tail 0
AT< chksum 0x63
AT< load 0x3ffe8000, len 776, room 8 
AT< tail 0
AT< chksum 0x02
AT< load 0x3ffe8310, len 552, room 8 
AT< tail 0
AT< chksum 0x79
AT< csum 0x79
AT< 
AT< 2nd boot version : 1.5
AT<   SPI Speed      : 40MHz
AT<   SPI Mode       : DIO
AT<   SPI Flash Size & Map: 8Mbit(512KB+512KB)
AT< jump to run user1 @ 1000
AT< 
AT< rlAT= ready
AT> AT+CWMODE=3
AT< 
AT< AT+CWMODE=3
AT< 
AT= OK
AT> AT+CIPMUX=1
AT< 
AT< AT+CIPMUX=1
AT< 
AT= OK
AT> AT+CWDHCP=1,1
AT< 
AT< AT+CWDHCP=1,1
AT< 
AT= OK
AT> AT+CWJAP="janUK","XXXX"
AT< 
AT< AT+CWJAP="janUK","XXXX"
AT< WIFI CONNECTED
AT< WIFI GOT IP
AT< 
AT= OK
AT> AT+CIFSR
AT< 
AT< AT+CIFSR
AT< +CIFSR:APIP,"192.168.4.1"
AT< +CIFSR:APMAC,"5e:cf:7f:14:0d:ee"
AT= +CIFSR:STAIP,"192.168.43.208"
AT< 
AT< +CIFSR:STAMAC,"5c:cf:7f:14:0d:ee"
AT< 
AT= OK
[EasyConnect] Connected to Network successfully
AT> AT+CIFSR
AT< 
AT< AT+CIFSR
AT< +CIFSR:APIP,"192.168.4.1"
AT< +CIFSR:APMAC,"5e:cf:7f:14:0d:ee"
AT= +CIFSR:STAIP,"192.168.43.208"
AT< 
AT< +CIFSR:STAMAC,"5c:cf:7f:14:0d:ee"
AT< 
AT= OK
[EasyConnect] IP address 192.168.43.208
[SMC] Device name ba7731fe-ec2f-4a73-9c63-554edfd4103b
AT> AT+CIPSTART=0,"UDP","8.8.8.8",53
AT< 
AT< AT+CIPSTART=0,"UDP","8.8.8.8",53
AT< 0,CONNECT
AT< 
AT= OK
AT> AT+CIPSEND=0,40
AT< 
AT< AT+CIPSEND=0,40
AT< 
AT< OK
AT= >
AT< 
AT< Recv 40 bytes
AT< 
AT< SEND OK
AT< 
AT! +IPD
AT= ,0,56:
```

After this: nothing. On an ancient version of the Espressif firmware it does work.

geky, 2016-08-30:
When we were designing the esp8266-driver, we decided to choose a single firmware and stick with it. Supporting different firmwares would be nice, but difficult to test against and retain stability.

That being said, with #9 merged, do you think we should move from the Espressif 1.5 SDK to the Espressif 2.0 SDK and update the examples and flashing instructions we have on the developer site?

Are there any concerns with adopting the 2.0 SDK? Do you by any chance know the size difference?

janjongboom, 2016-08-30:
@geky actually it does not currently work with 1.1, 1.5 or 2.0 SDK... (I tried the NOSDK releases for all of them). The only firmware that works is the firmware at https://developer.mbed.org/teams/ESP8266/code/ESP8266-Firmware-Update-To-Espressif/... But that is not flashable to all modules. 

We tried everything to get Wifi3 Click module to work, so upgraded them to all three (1.1, 1.5 and 2.0) and none worked... So I think the fw that we had published is some very peculiar one...

geky, 2016-08-30:
Ah, my bad on the mistaken version. Looking into it more, it appears we are using the 0.2 SDK, which is a little bit old.

We should definitely adopt a more recent version. Will need to update flashing instructions appropriately.

geky, 2017-05-09:
With https://github.com/ARMmbed/esp8266-driver/pull/21, looks like this issue is all good now : )

closed 2017-05-09 by geky