ARMmbed/esp8266-driver #12: Doesn't compile against mbed-os master

tommikas, 2016-10-03:
```
[ERROR] In file included from ./esp8266-driver/ESP8266Interface.cpp:18:0:
./esp8266-driver/ESP8266Interface.h:20:41: fatal error: network-socket/NetworkStack.h: No such file or directory
 #include "network-socket/NetworkStack.h"
```

Should be `netsock/NetworkStack.h` now.

geky, 2016-10-03:
Thanks for reporting this. However, there is another set of changes staged for the WiFiInterface that will deeply impact the implementation of the esp8266-driver: https://github.com/ARMmbed/mbed-os/pull/2874

Also there is the concern of breaking the version used for the release (which has been problematic https://github.com/ARMmbed/esp8266-driver/issues/7). Perhaps this repo needs a dev branch?

tommikas, 2016-10-04:
>Perhaps this repo needs a dev branch?

Whatever works the best. We need to be able to use the driver with mbed-os master, so as long as that's possible it's fine.

Speaking of incoming changes, are the currently [unsupported socket operations](https://github.com/ARMmbed/esp8266-driver/blob/master/ESP8266Interface.cpp#L134-L137) going to be implemented at some point?

geky, 2016-10-04:
The unsupported operations were actually due to technical limitations in the 0.2 version of the esp8266 firmware. Support requires a bit of effort in updating to 2.0 but is planned.

tommikas, 2016-10-04:
I figured there might have been something like that. Nice to hear it's coming :)

tommikas, 2016-10-05:
Closing, since this has now been fixed.

closed 2016-10-05 by tommikas