ARMmbed/esp8266-driver #14: Esp8266 error

meyucel, 2016-10-18:
Error: Identifier "WiFiAccessPoint" is undefined in "ESP8266Interface/ESP8266/ESP8266.h", Line: 118, Col: 15
Error: Identifier "nsapi_wifi_ap_t" is undefined in "ESP8266Interface/ESP8266/ESP8266.h", Line: 205, Col: 19
Error: Identifier "WiFiAccessPoint" is undefined in "ESP8266Interface/ESP8266/ESP8266.cpp", Line: 148, Col: 20
Error: Identifier "nsapi_wifi_ap_t" is undefined in "ESP8266Interface/ESP8266/ESP8266.cpp", Line: 151, Col: 6
Error: Identifier "WiFiAccessPoint" is undefined in "ESP8266Interface/ESP8266/ESP8266.cpp", Line: 159, Col: 25
Error: Identifier "nsapi_wifi_ap_t" is undefined in "ESP8266Interface/ESP8266/ESP8266.cpp", Line: 296, Col: 24
Error: Identifier "NSAPI_SECURITY_UNKNOWN" is undefined in "ESP8266Interface/ESP8266/ESP8266.cpp", Line: 303, Col: 55

0xc0170, 2016-10-18:
@meyucel What is the version of mbed-os? 

Please provide more details, to be able to reproduce it. I would assume you are using older version of mbed-os.

meyucel, 2016-10-18:
Hi Martin @0xc0170,

I used the latest version, I think it is mbed-os-5.2.0-rc3. I used the auto update tool, revision number is 2246:a6f3fd1. In addition to the above problem, I import the example in https://github.com/ARMmbed/oob-mbed-os-example-wifi. There is no problem and it compiles as usual. However when I update the mbed-os to lastest version, it gives same errors too.

Thanks in advance.

meyucel, 2016-10-18:
I also noticed that if I try to use the Serial connection to PC,  it cannot send anything with this lib. I think this is beacuse buffered serial code inside lib.

I get these warning

`Warning: Function "mbed::SerialBase::attach(T *, void (T::*)(), mbed::SerialBase::IrqType) [with T=BufferedSerial]" (declared at <a href="#" onmousedown="mbed_doc_goto('/oob-mbed-os-example-wifi//extras/mbed-os.lib/drivers/SerialBase.h', '117'); return false;">/extras/mbed-os.lib/drivers/SerialBase.h:117</a>) was declared deprecated ("The attach function does not support cv-qualifiers. Replaced by attach(callback(obj, method), type). [since mbed-os-5.1]") in "esp8266-driver/ESP8266/ATParser/BufferedSerial/BufferedSerial.cpp", Line: 31, Col: 17
`

0xc0170, 2018-06-13:
This refers to old documentation page that is deprecated. This should be fixed already.

closed 2018-06-13 by 0xc0170