ARMmbed/esp8266-driver #15: recvfrom does not return valid SocketAddress

geky, 2016-11-23:
Currently, the recvfrom for UDP is just a thin wrapper over recv, because of this no valid SocketAddress is returned to the application.

Either the command set needs to be updated to obtain the recvfrom IP address, or the IP address of the connection should be stored for future recvfroms.

jankii01, 2016-11-30:
This would really need fixing, seems provisioning is REALLY checking this and now it fails...

geky, 2016-12-02:
Hi @jankii01, with #16 merged, this should be fixed as best as possible with the current firmware. Let us know if there is there an issue.

jankii01, 2016-12-02:
Thanks! I've started a test run with that PR in.

jankii01, 2016-12-02:
https://github.com/ARMmbed/mbed-cloud-client-example/pull/73 this PR of course in reality.

closed 2017-05-09 by geky