ARMmbed/esp8266-driver #17: vsend of class ATParser only sends one character with active uVisor

Iceberg1988, 2016-12-03:
I tryed to get the ESP library working with active uVisor, but it stuck after sending a reset to the module. During measurements on the portpins of the MCU I only saw one single character gets out. So I started to debug it a bit:
By a call of `_parser.send("AT+RST")` the method vsend seems to write characters properly by calling putc() inside the loop of line [#L168](https://github.com/ARMmbed/esp8266-driver/blob/master/ESP8266/ATParser/ATParser.cpp#L168). But on the hardware I only can measure the first sign `A`.
During stepping through the code I observed the loop and found out, that the characters apears one by one on the wire if I step through this loop. Maybe this is some kind of timing problem and the MPU isn't fast enought for the loop and the buffered serial throws the other characters?

I tested this with ESP8266-driver at [commit dc37b6](https://github.com/ARMmbed/ESP8266-driver/commit/dc37b65ca877d969aa492f348626df6e1b0b1df0) and [commit 86ed47](https://github.com/ARMmbed/ESP8266-driver/commit/86ed47c8bf40c86fdc569c7e678a8944f5e01c2a)
used mbed-os is at [commit a1c084](https://github.com/ARMmbed/mbed-os/commit/a1c0840b3d69060e5eb708edb18358e424a40f51)
and uVisor is at version is v0.25.1

My ESP8266 module and setup (connection to uart, power supply...) works properly without active uVisor.

meriac, 2016-12-09:
@Iceberg1988 my first hunch is that the IRQ is not registered. Keep in mind that uVisor expects NVIC_etInterrupt to be used - and ignores link time interrupt vectors.

cmonr, 2018-06-11:
Closing due to old age.

closed 2018-06-11 by cmonr