ARMmbed/esp8266-driver #18: Esp8266 doesn't send/receive packets with size above 1500 bytes

shelib01, 2016-12-04:
I run Device Identity flow using WiFI interface (esp8266 driver). We have some big certificate which should arrive from Identity Server. It's network packet  size is relatively big (above 1500 bytes). This packet is disappeared. All over packets were sent/received successfully.
The same code runs perfect with Ethernet interface.

jankii01, 2016-12-13:
NOTE! This does not apply to mbed-os-example-client.

kjbracey-arm, 2016-12-13:
You can assume a /little/ more than those quotes - nodes should(IPv4)/must(IPv6) be able to reassemble packets up to the size of the link they're coming from. So an Ethernet-connected IPv4 node can reasonably be expected to reassemble up to 1500 rather than just 576.

(An Ethernet host must obviously be able to handle 1500-byte packets if unfragmented, and you don't want an intermediate small tunnel between two Ethernet hosts fragmenting traffic to totally break a protocol that would have worked if it was just Ethernet. Although the fragmentation will still cause performance issues.) 

In theory you should play it safe and assume 576 for remote hosts, both for the benefit of the end device, and to avoid performance-sapping intermediate fragmentation. In practice you may be able to go higher if you are prepared for fragmentation and can assume that the remote end is Ethernet-based so can handle 1500, but >1500 is right out and will fail on a large variety of real systems, just as it's failing here.

Most basic protocols do this:

* DNS - 512 bytes UDP payload
* TFTP - 512 bytes UDP payload.
* TCP - 536 bytes IP payload unless other end says more (and our end is sophisticated enough to have path MTU discovery to avoid IP fragmentation - or maybe if we spot we're on the same net). lwIP just uses 536.

Aside from the formal standards point of view above, there is the simple practical matter that we're dealing with constrained systems here. If I was configuring a constrained system, one of the first things I'd do was make sure it was using the minimum amount of buffer RAM I could get away with in the IP stack. On a host using something smaller than Ethernet I would not want to be reserving 1500 (let alone more) when the IP spec is clear I don't need to, depriving the applications of much-needed RAM.

Even if you do have a system that's in principle capable of IP-reassembling more than the minimum, forcing the data through as a single datagram greatly increases total memory usage, as you are forcing the stack to have space for reassembly, then after reassembly you need enough RAM allocated simultaneously for the datagram in the network stack and in the receive buffer for the application. When you make the `recv(buffer, big_data_size)` call for that datagram, peak memory usage for just that call is 2 * big_data_size.

In the ESP8266's case it would be forcing it to hold the entire data in its RAM before passing it all in one go to the main CPU. I'm not sure what it's memory limits are, but it's not being at all friendly to it.

Using something like TFTP or blockwise CoAP or TCP gives you the flexibility to handle unlimited amounts of data without worrying about memory constraints, or ever being surprised by the next system that doesn't quite have enough RAM, or when the payload gets just a little big bigger still, blowing currently-working systems' reassembly buffer size.

I note the existing mbed-client does allow resources to receive a "PUT" bigger than the system's entire RAM, using CoAP blockwise transfers - the application is notified for each block received as they arrive, and the mbed-client/CoAP system doesn't try to hold the entire data.

TeroJaasko, 2016-12-15:
I was just wondering if the driver API should have the MTU defined somewhere for client to query? Or should/will the socket_send() functions return failure if one tried to send too big datagram for driver to handle?

kjbracey-arm, 2016-12-15:
It's the receiving that's the problem, not the sending. The fragments of the large UDP datagram make it across the network, but the receiver fails to reassemble it, and drops silently (as is permitted).

Reading the link MTU at the transmitter end doesn't necessarily give you a size that the receiver can handle (unless you know you're on the same link).

If there was an API to read either the IP maximum reception size or link MTU at the receiver's end, you could communicate it to the transmitter as part of your handshake protocol (like TCP's MSS option). But that still presupposes having the ability to actually segmentise transfers at the transmitter's end.

And once you are segmentising, rather than just querying "could the receiver handle this segment size", anything that can segmentise should ideally aim for a smaller size that avoids IP fragmentation, but optimising that rather than just using the conservative 576/1280 involves some fiddly IP stack bits and slightly advanced path MTU socket API.  ( https://tools.ietf.org/html/rfc3542#page-44 for IPv6 ).

On the error point - generally send() will accept any output size if fragmentation is enabled (as it should be by default). Fragment output is usually less constrained than input, as streaming out the fragments of a large datagram is easier than reassembling bits.

If you do disable send fragmentation (via advanced socket options), that RFC notes that implementations may not able to generate "EMSGSIZE" errors - they could be quietly dropped.

Good general fragmentation discussion here: https://labs.ripe.net/Members/gih/evaluating-ipv4-and-ipv6-packet-fragmentation

cmonr, 2018-06-11:
Closing due to inactivity.

closed 2018-06-11 by cmonr