ARMmbed/esp8266-driver #22: ESP8266 does not return '0' on closed socket

janjongboom, 2017-04-06:
According to TCPSocket spec calling `recv()` on a closed socket should return `0`.

When you do this on esp8266-driver the socket hangs forever.

Reproduce:

1. Find a server which closes the connection directly after the response (e.g. Philips Hue hub).
1. Make a request (`send()`).
1. Call `recv()` a few times, e.g. via:

    ```cpp
    socket.recv(buffer, 1024);
    socket.recv(buffer, 1024);
    socket.recv(buffer, 1024);
   ```

At some point you'll see:

```
AT< 
AT< SEND OK
AT< 
AT! +IPD
AT= ,0,17:
AT< 
AT= OK
recv1 17 <--- first packet, is OK!
AT< 
AT< 
AT! +IPD
AT= ,0,481:
AT< 
AT= OK
recv2 481  <--- second packet is OK too!
AT< 
AT< 0,CLOSED <---- socket is closed
AT< 
AT< ERROR <---- third recv() call hangs forever now
```

Now the whole app hangs.

cmonr, 2018-06-11:
Closing due to inactivity.

janjongboom, 2018-06-18:
@cmonr Why are we closing due to inactivity? This is a remaining bug.

kjbracey-arm, 2018-06-18:
FWIW, there has been quite a bit of work in this area on the driver in the last year. I would expect this to have been fixed by https://github.com/ARMmbed/esp8266-driver/pull/60

janjongboom, 2018-06-19:
@kjbracey-arm Check, thanks.

closed 2018-06-19 by janjongboom