ARMmbed/esp8266-driver #25: Failure in sending UDP packets

hasnainvirk, 2017-04-26:
_parser->send("AT+CIPSENDBUF=%d,%d", id, amount) fails.

Probably the firmware in the chip have a bug as previously we had seen a failure with CIPDOMAIN command which got resolved by updating the chip firmware to the latest.

@geky @sarahmarshy

kegilbert, 2017-04-26:
@sarahmarshy 
Tested with this line reverted: https://github.com/ARMmbed/esp8266-driver/commit/1240d829f74657234ccb6a2314d3604017569be7#diff-87361473d6d94be073cd39f6275e2325L186 on the udp_echo test modified to run on the ESP8266. 

It failed with master, and passes with that line reverted. CIPSENDBUF looks like it's for TCP connections. Was there a reason to switch to that command? Not really familiar with the backstory behind that commit so deferring to you for more info.

BlackstoneEngineering, 2017-04-27:
All, please reference which version of both the driver and the firmware on the ESP8266 you are using. This chip has many different firmware versions, each of which use slightly different AT Command sets. We officially support only one firmware package and have recently updated (last week), so please re-confirm issue with the firmware version. 

Updated details can be found here : https://developer.mbed.org/components/ESP8266-01/ with firmware update instructions here : https://developer.mbed.org/teams/ESP8266/wiki/Firmware-Update

hasnainvirk, 2017-04-27:
I  have version esp8266_v2.0

kegilbert, 2017-04-27:
Looks to be version 2 on our end as well.

geky, 2017-04-27:
I went ahead and merged @kegilbert's pr since #24 is blocked by the related pr on mbed-os: https://github.com/ARMmbed/mbed-os/pull/4119

I think this issue is no longer present, so I'll go ahead and close it, feel free to reopen if I'm wrong

closed 2017-04-27 by geky