ARMmbed/esp8266-driver #33: Some tests fail on K64F & ESP8x

MarceloSalazar, 2017-05-19:
Daplink v243
mbed CLI 1.1.1
mbedgt 1.2.5
Windows 7
Both PC and ESP8x connected to same local wifi network

esp8266-driver sha:95908fbaa06bcf32dcd196edc85ffc788c6f48c7
mbed-os sha: 5d0ce3c53158c69b5456d1d2ef4ecc12fdcfc49e

Test summary
```
+--------------+---------------+------------------------------+----------------------------------------+--------+--------+--------+--------------------+
| target       | platform_name | test suite                   | test case                              | passed | failed | result | elapsed_time (sec) |
+--------------+---------------+------------------------------+----------------------------------------+--------+--------+--------+--------------------+
| K64F-GCC_ARM | K64F          | tests-net-connectivity       | Bringing the network up and down       | 1      | 0      | OK     | 5.5                |
| K64F-GCC_ARM | K64F          | tests-net-connectivity       | Bringing the network up and down twice | 1      | 0      | OK     | 10.96              |
| K64F-GCC_ARM | K64F          | tests-net-gethostbyname      | DNS literal                            | 1      | 0      | OK     | 0.09               |
| K64F-GCC_ARM | K64F          | tests-net-gethostbyname      | DNS preference literal                 | 1      | 0      | OK     | 0.1                |
| K64F-GCC_ARM | K64F          | tests-net-gethostbyname      | DNS preference query                   | 1      | 0      | OK     | 0.24               |
| K64F-GCC_ARM | K64F          | tests-net-gethostbyname      | DNS query                              | 1      | 0      | OK     | 0.21               |
| K64F-GCC_ARM | K64F          | tests-net-tcp_echo           | TCP echo                               | 0      | 1      | FAIL   | 6.13               |
| K64F-GCC_ARM | K64F          | tests-net-tcp_hello_world    | TCP hello world                        | 1      | 0      | OK     | 7.52               |
| K64F-GCC_ARM | K64F          | tests-net-udp_dtls_handshake | UDP DTLS handshake                     | 0      | 1      | FAIL   | 36.83              |
| K64F-GCC_ARM | K64F          | tests-net-udp_echo           | UDP echo                               | 0      | 0      | ERROR  | 0.0                |
+--------------+---------------+------------------------------+----------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 2 FAIL / 7 OK / 1 ERROR
```

@geky What would be the reason for this?
Could it be a timing issue in the driver or the tests?

geky, 2017-05-19:
Hi @MarceloSalazar, thanks for the test log, that helps a bunch.

It looks like the esp8266 can't establish a connection to your host PC. For those three tests, the esp8266 is supposed to connect to your computer, but the logs are showing that you can't.

Is it possible that there is something preventing the device and PC from talking to each other? Maybe your firewall is getting in the way?

MarceloSalazar, 2017-05-22:
Thanks @geky, you seem to be right, I've moved to a PC with Windows 10 and things have improve a bit. All tests usually pass (w/wo DEBUG), but sometimes random tests fail.
See log with lots of tests:

I don't know whether this might be caused by the ESP8x fw, or the tests itself.

Also, the connectivity test sometimes fails when running it alone multiple times (with DEBUG enabled). See log.

geky, 2017-05-24:
Ah! that's good new. In the new results, it looks like the esp8266 just isn't able to connect to the wifi occasionally. We can add some retry logic to the tests for this.

geky, 2017-05-24:
Hi @MarceloSalazar, are you able to try this branch and see if it still shows the intermitent failures?
https://github.com/ARMmbed/esp8266-driver/tree/g-connect-retries

MarceloSalazar, 2017-05-31:
@geky I've been running the tests based on your branch the whole day.
The tests run in the morning are very unreliable (half of them fail), whether the ones run in the afternoon consistently pass all the time (more than 20 cycles completed!).
I suspect this might be related to the wifi network infrastructure, but cannot really say.
I'd be happy if you merge the branch and then close this issue.

geky, 2017-06-05:
I don't actually want to merge https://github.com/ARMmbed/esp8266-driver/tree/g-connect-retries

I'd consider retries at this level to be an issue with the test framework, so I'll move them there.

cmonr, 2018-06-11:
Closing due to inactivity.

closed 2018-06-11 by cmonr