ARMmbed/esp8266-driver #36: DNS request of a hostname causes device to become unresponsive

moonchen, 2017-07-06:
After performing a gethostbyname on a domain name that is long, such as "firmware-catalog-media-ca57.s3.dualstack.us-east-1.amazonaws.com", or "3.141592653589793238462643383279502884197169399375105820974944592.com", the ESP8266 stops responding.

Test case: modify line 12 of tests/net/gethostbyname/main.cpp to read
#define MBED_DNS_TEST_HOST "firmware-catalog-media-ca57.s3.dualstack.us-east-1.amazonaws.com", then run the test case.  This causes the DNS query test to fail.

JanneKiiskila, 2017-07-07:
@sg-  - potential OSCA case, please label.

imsherlock, 2017-07-07:
Just a quick update on this since I'm working with Mo. I've got an ESP8266 WiFi module running the following firmware

AT+GMR
**AT version:1.4.0.0(May  5 2017 16:10:59)
SDK version:2.1.0(116b762)
compile time:May  5 2017 16:37:51**

I've issued the AT+CIPDOMAIN as follows to lookup the URL mentioned above.

AT+CIPDOMAIN="firmware-catalog-media-ca57.s3.dualstack.us-east-1.amazonaws.com"
**DNS Fail

ERROR**

Using http://simpledns.com/lookup-dg.aspx I'm able to trace the DNS lookup to see that firmware-catalog-media... is a CNAME record for s3.dualstack.us-east-1.amazonaws.com. Performing the same command with this URL works

AT+CIPDOMAIN="s3.dualstack.us-east-1.amazonaws.com"
**+CIPDOMAIN:52.216.0.123

OK**

Our initial thought was that this may be a character limitation and I think this can be confirmed by changing the URL given by Mo with -ca1; this reduces the number of characters in the string from 64 to 63. You can verify that this is a valid DNS record and another CNAME for s3.dualstack.us-east-1.amazonaws.com using the tool above. Using the AT+CIPDOMAIN command works

AT+CIPDOMAIN="firmware-catalog-media-ca1.s3.dualstack.us-east-1.amazonaws.com"
**+CIPDOMAIN:54.231.72.106

OK**

The DNS specification says that labels may be up to 63 characters and the full domain name is limited to 253 characters. None of the labels in the domain name above are more than 63 characters but it seems like this may be the limitation we are running into.

geky, 2017-07-07:
Interesting problem, and thanks all of the info. Since the update to the ESP8266 firmware (https://github.com/ARMmbed/esp8266-driver/commit/1240d829f74657234ccb6a2314d3604017569be7), we've just handed off DNS queries to the CIPDOMAIN command. Unfortunately we don't have the source on the ESP8266 so I can only guess that they have an internal limitation.

You could try falling back to mbed's DNS implementation by removing the `gethostbyname` function:
https://github.com/ARMmbed/esp8266-driver/blob/master/ESP8266Interface.h#L73

That will fall back to the default implementation in mbed, which should only be limited by the max size of a DNS packet.

geky, 2017-07-18:
I think this was fixed with https://github.com/ARMmbed/esp8266-driver/pull/38

Feel free to reopen

closed 2017-07-18 by geky