ARMmbed/esp8266-driver #39: [NUC472] Fail to connect to device connector

ccli8, 2017-07-25:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_NUC472

**Version information**
mbed-os-example-client (mbed-os-5.5.2)
```
mbed-os-example-client (868f1585f40e)
|- easy-connect (cc471beb7e26)
|  |- atmel-rf-driver (0ae76ce17ae5)
|  |- esp8266-driver (918caa01b4cd)
|  |  `- ESP8266\ATParser (269f14532b98)
|  |- mcr20a-rf-driver (d5905fefa54c)
|  `- stm-spirit1-rf-driver (09e4b9664de1)
|- mbed-client (31e5ce203cc0)
|  |- mbed-client-c (ecfa619e42b2)
|  |- mbed-client-classic (4e66929607c3)
|  `- mbed-client-mbed-tls (7e1b6d815038)
|- mbed-os (8828635da469)
`- pal (60ce64d5ec35)
```

**Steps to Reproduce**
Without modification, K64F can connect to mDC but NUC472 cannot.

<pre>
Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (ESP8266) 
[EasyConnect] Connecting to WiFi VVVVVVVVVV
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address a0:20:a6:09:95:db
[EasyConnect] IP address 192.168.43.196

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered
simulate button_click, device not registered

<b>[ERROR:] M2MInterface::SecureConnectionFailed</b>
</pre>
With further check into ESP8266 driver, there is mismatch between driver and AT command. According to **+IPD - Receives Network Data** command in [AT command set](https://www.espressif.com/sites/default/files/documentation/4a-esp8266_at_instruction_set_en.pdf), there is no "OK" in response. So `_parser.recv("OK")` always returns `false` and socket layer needs another `ESP8266::recv` call to get packet data. If this happens as a complete packet has received by driver, there will be no further interrupt to inform socket layer and this `ESP8266::recv` call is the last one. That's why **NUC472** fails to connect to mDC.
<pre>
int32_t ESP8266::recv(int id, void *data, uint32_t amount)
{
    while (true) {
        // check if any packets are ready for us
        for (struct packet **p = &_packets; *p; p = &(*p)->next) {
            if ((*p)->id == id) {
                struct packet *q = *p;

                if (q->len <= amount) { // Return and remove full packet
                    memcpy(data, q+1, q->len);

                    if (_packets_end == &(*p)->next) {
                        _packets_end = p;
                    }
                    *p = (*p)->next;

                    uint32_t len = q->len;
                    free(q);
                    return len;
                } else { // return only partial packet
                    memcpy(data, q+1, amount);

                    q->len -= amount;
                    memmove(q+1, (uint8_t*)(q+1) + amount, q->len);

                    return amount;
                }
            }
        }

        // Wait for inbound packet
        <b>if (!_parser.recv("OK")) {</b>
            return -1;
        }
    }
}
</pre>
If I make the following modification,
<pre>
int32_t ESP8266::recv(int id, void *data, uint32_t amount)
{
    while (true) {
        ...
        // Wait for inbound packet
        if (!_parser.recv("OK")) {
            
           <b>
            // MY FIXUP
            struct packet **p = &_packets;
            if (*p && (*p)->id == id) {
                continue;
            }
            </b>
            
            return -1;
        }
    }
}
</pre>
**NUC472** can connect to mDC successfully.
<pre>
Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (ESP8266) 
[EasyConnect] Connecting to WiFi VVVVVVVVVV
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address a0:20:a6:09:95:db
[EasyConnect] IP address 192.168.43.196

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684
simulate button_click, device not registered

<b>Registered object successfully!</b>
simulate button_click, new value of counter is 1
simulate button_click, new value of counter is 2
</pre>

ccli8, 2017-08-24:
Hi, any update?

yogpan01, 2017-08-25:
@geky Can you please check this ?

ccli8, 2017-11-09:
I re-verify it OK on esp8266-driver (https://github.com/ARMmbed/esp8266-driver/#fdb44a4bedbb5879a8ff9afd864e296765c20f22) and ATParser (https://github.com/ARMmbed/ATParser/#a2cb09990f11a17504608272966f1185b211ce23). It should be fixed in some version.

closed 2017-11-09 by ccli8