ARMmbed/esp8266-driver #42: IPD data is not separated per socket when two simultaneous requests come in

janjongboom, 2017-07-31:
If you have two simultaneous requests come in (e.g. Firefox requesting both `/` and `/favicon.ico`, the data from both requests gets mangled. Maybe add a mutex around reading the IPD data...

F.e. here's the data I get back on one of my sockets:

```
GET / HTTP/1.1
Host: 192.168.1.12:8080
User-AgenGecko/20100101 Firefox/56.0
Accept: text/html,application/xhtml+xml,Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
Connection: keep-alive
Upgrasts: 1
Pragma: no-cache
Cache-Control: no-cache
```

While the request actually was:

```
GET / HTTP/1.1
Host: 192.168.1.12:8080
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.11; rv:56.0) Gecko/20100101 Firefox/56.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
Connection: keep-alive
Upgrade-Insecure-Requests: 1
Pragma: no-cache
Cache-Control: no-cache
```

janjongboom, 2017-07-31:
So it actually comes over the line in order, but bytes are missing from ATParser...

janjongboom, 2017-07-31:
Patched this by using raw IRQ in my server examples, maybe adding the IRQ handling actually made the problem happen (because of `malloc`) but not sure...

SeppoTakalo, 2018-06-13:
This has already been fixed by using UARTSerial that has internal buffering and mutexes in place.

closed 2018-06-13 by SeppoTakalo