ARMmbed/esp8266-driver #44: Question: esp8266 doesn't respond reset command "AT+RST"

soramame21, 2017-09-02:
I got some failure after bootloader starts downloaded firmware through wifi.

geky, 2017-09-03:
Unfortunately, the esp8266 can just end up in some bad states where it won't parse AT+RST command. This is mostly caused by resetting the microcontroller.

Best example: You send AT+SEND 4096 bytes, but the microcontroller is reset halfway through sending the data. Now you would have to send AT+RST 600 times before the esp8266 starts parsing commands again. You will also send a TCP packet as a side-effect, which may be problematic upstream.

The best solution I can think of is to tie the reset lines together. Some packages have the reset line exposed, some just have a reset button. Note, reset lines will need to be untied for firmware updates.

There have been talks of adding a pin argument for the reset line, but no one has had the time to add it yet.

soramame21, 2017-09-04:
@geky Thanks for explanation. If we can reset the esp8266 shield after update, that will be perfect.

closed 2017-09-04 by soramame21