ARMmbed/esp8266-driver #48: Failing to connect with No MAC Address

teetak01, 2017-10-23:
Using easy-connect with Nucleo F429ZI + ESP8266:

```
00:03:53.750 [EasyConnect] IPv4 mode
00:03:53.750 [EasyConnect] Using WiFi (ESP8266)
00:03:53.750 [EasyConnect] Connecting to WiFi systest
00:03:53.750 [EasyConnect] ERROR - No MAC address
00:03:53.750 [EasyConnect] Connection to Network Failed -3004!
```

This is rare event, but seems to be some issue with the driver.

geky, 2017-10-23:
If the wifi only sometimes connect, it's probably noise on the wifi channel. The only correct behaviour is to retry the connection.
see also https://github.com/ARMmbed/esp8266-driver/issues/33

We could add a "retry" option to handle this driver side, though I'm not sure when we'll be able to get to it. I went ahead and created a jira ticket so it is tracked [IOTMORF-1638](https://jira.arm.com/browse/IOTMORF-1638)

SeppoTakalo, 2018-06-13:
This has already been fixed by moving the MAC address printing to happen only if connection is succesfull: https://github.com/ARMmbed/easy-connect/blob/master/easy-connect.cpp#L285

closed 2018-06-13 by SeppoTakalo