ARMmbed/esp8266-driver #54: Why ESP8266 interface disable soft AP mode

cyliangtw, 2017-12-26:
Normally, WiFi consumer device could support soft AP mode for user setting through smart phone or NB.
Moreover, ESP8266 firmware also could support WiFi soft AP mode by AT command.
Why ESP8266 interface not prefer to support this soft AP mode ?

SeppoTakalo, 2018-01-08:
Soft AP mode has never been working with this device.
Having it enabled just makes the device broadcast its AP name without any usable functionality.

Also, if we were to enable it, it should be configurable by user whether it is enabled or not.

closed 2018-01-08 by SeppoTakalo

cyliangtw, 2018-01-10:
@SeppoTakalo  , ESP8266 device should be ready for softAP, ESP8266 SDK release_note mentioned optimization of softAp severla times and Arduino also guide how to manipulate ESP8266 softAp.
   http://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/soft-access-point-class.html
So, is it doable to support the similar softAp control on this device in Mbed ?

SeppoTakalo, 2018-01-10:
Yes, true.

ESP8266 SDK and Arduino's ESP8266 drivers support Soft AP mode.
We don't.

There is no equivalent APIs in Mbed OS so this driver does not implement any extra.

In order to support soft AP mode, it not just to turn it on by AT commands, the support has to be developed in the driver also. Currently we don't have it, and therefore the mode has been turned off in the device.
Previously the SoftAP mode was enabled by AT commands, which was a error.

If the SoftAP is requested feature, we need to create another task of it and develop&test it. But this issue was about "why it was disabled" and the answer is because it is unsupported.

cyliangtw, 2018-01-10:
@SeppoTakalo , thanks of your explanation.
Actually, I have some customer demand on WiFi softAp mode for setting.
Should I create another one issue for "enquiry softAp" or just control it  through proprietary WiFi AT command ?