ARMmbed/esp8266-driver #88: status update missing

lauri-piikivi, 2018-09-17:
https://github.com/ARMmbed/esp8266-driver/blob/7bb1d8e27be54f9f92805a38176478d795aa9dfb/ESP8266/ESP8266.cpp#L675

should this be handled as status update to app, and reconnect logic?

lauri-piikivi, 2018-09-17:
meant the function above "ESP8266::_connect_error_handler()" but if socket error can not be pinpointed to specific socket, should the error handling be for all, ie network interface?

VeijoPesonen, 2018-09-18:
If UNLINK ERROR is received when socket is closed by the app we know ESP8266 has closed the socket from its side prematurely. It has a bad habit of acting on its own.
https://github.com/ARMmbed/esp8266-driver/blob/7bb1d8e27be54f9f92805a38176478d795aa9dfb/ESP8266/ESP8266.cpp#L594

closed 2018-10-02 by adbridge